ARISE-Initiative/robomimic #217: Attempting to test installation

neelbg01, 2025-02-21:
Hey there,

I have been attempting to test my installation to make sure it works however I running into a few issues where example scripts are missing 

```
simple-config-example: maybe failed - stderr output below (if it's only from tqdm, the test passed)
python: can't open file '/home/***/anaconda3/lib/python3.12/site-packages/robomimic/../examples/simple_config.py': [Errno 2] No such file or directory

simple-obs-nets-example: maybe failed - stderr output below (if it's only from tqdm, the test passed)
python: can't open file '/home/***/anaconda3/lib/python3.12/site-packages/robomimic/../examples/simple_obs_nets.py': [Errno 2] No such file or directory

simple-train-loop-example: maybe failed - stderr output below (if it's only from tqdm, the test passed)
python: can't open file '/home/***/anaconda3/lib/python3.12/site-packages/robomimic/../examples/simple_train_loop.py': [Errno 2] No such file or directory

train-bc-rnn-example: maybe failed - stderr output below (if it's only from tqdm, the test passed)
python: can't open file '/home/***/anaconda3/lib/python3.12/site-packages/robomimic/../examples/train_bc_rnn.py': [Errno 2] No such file or directory
```

Has anyone else run into this issue?

Regards,
Neel

amandlek, 2025-02-21:
Which version and branch of robomimic and robosuite are you on?

michelleezhang, 2025-03-01:
I'm not sure about the missing example scripts (my guess is you might have accidentally missed a step during installation), but I think the OSC_POSE controller assertion error is due to the robomimic datasets being incompatible with robosuite v1.5 -- if you are using the latest version of robosuite, then downgrading to robosuite v1.4.1 should fix this!

michelleezhang, 2025-03-01:
I'm not sure about the missing example scripts, but I think the OSC_POSE controller assertion error is due to the robomimic datasets being incompatible with robosuite v1.5 -- if you are using the latest version of robosuite, then downgrading to robosuite v1.4.1 should fix this!

amandlek, 2025-03-11:
Closing, as the latest master branch should now work with robosuite v1.5

closed 2025-03-11 by amandlek