ARISE-Initiative/robomimic #221: Is it possible to create my own camera when trying to convert states dataset to obs dataset?

zaizailu, 2025-03-21:
I want to use the trajs from your datasets, like robomimic v0.1. I would like to use cameras with my own camera configs to render the obs. Is there any way to do so?

amandlek, 2025-03-21:
Great question! You would have to edit the model xml in each demonstration to add your new camera, before resetting the environment with the model xml for observation extraction. 

To be more concrete, take the model xml [here](https://github.com/ARISE-Initiative/robomimic/blob/1203cd2192d409070239a9ce06b33e2a64298157/robomimic/scripts/dataset_states_to_obs.py#L264) and then edit it to insert the camera(s) you'd like. For a rough example of how you might do this, you can see [this](https://github.com/robocasa/robocasa/blob/d1dd1eb627d3f808decef0fc8bc5ab5bad997664/robocasa/environments/kitchen/kitchen.py#L1006-L1029).

Then you can run that normal observation extraction script and pass your new camera name(s) as arguments.

You can read more on observation extraction here: https://robomimic.github.io/docs/datasets/robosuite.html#extracting-observations-from-mujoco-states

Hope this is helpful!

closed 2025-03-21 by amandlek

zaizailu, 2025-03-21:
> Great question! You would have to edit the model xml in each demonstration to add your new camera, before resetting the environment with the model xml for observation extraction.
>
> To be more [...]

Thx, very helpful!