ARISE-Initiative/robomimic #226: How to run a trained policy in real time in simulation？

wudier483, 2025-04-06:
Hello, I would like to ask how to run a policy that requires camera observation in a simulation environment, that is, rendering on the screen, but the document saves the result as a video. I tried the following solution:

env = EnvUtils.create_env_from_metadata(
env_meta=env_meta,
render=True,
render_offscreen=False,
use_image_obs=True,
)

But an error message appears:
ValueError: the onscreen and offscreen renderers cannot be used simultaneously.

That is, rendering on the screen and using camera observation cannot be True at the same time, but my policy requires observing the image as input.

Is there any other way to visualize it? Maybe I overlooked something. I look forward to your reply.

amandlek, 2025-04-08:
Unfortunately, we can't guarantee that simultaneous on-screen and off-screen rendering works across platforms due to simulator limitations, but you could try commenting out the relevant assertion / exception, and seeing if it works fine on your platform for your use case.

Alternatively, I would suggest rendering the policy rollouts to a video - our default run_trained_agent.py supports this feature, as do normal policy evaluations that are run during robomimic training.

closed 2025-04-08 by amandlek

wudier483, 2025-04-14:
Thanks for your prompt reply. Can I only save the camera observation as the result? If so, when I use the dataset mentioned in the paper, the image observation resolution is only 84X84. How can I make the result clearer?

amandlek, 2025-04-15:
You could save the camera observations to a video, yes, but in some cases the resolution might be poor like you mentioned. Why don't you just run the agent and save the policy rollout videos? e.g. see [here](https://github.com/ARISE-Initiative/robomimic/blob/59ecd4048b91595b61cbcc2f3e4d9d666e4d4c04/robomimic/scripts/run_trained_agent.py#L323)