ARISE-Initiative/robomimic #230: Can´t get any successful evaluations since update to 0.4.0 and robosuite 1.5.1

PartyPenguin, 2025-04-13:
## Issue: Evaluation failures after robomimic 0.4.0 and robosuite 1.5.1 update

**Description:**

Since updating to robomimic version 0.4.0 and robosuite version 1.5.1, I am experiencing consistent evaluation failures across my trained models. This issue also appears to affect the evaluation of algorithms presented in the original robomimic paper when run with the updated versions. 

I have confirmed that I am using the latest dataset files as recommended for these versions. The training process itself seems to proceed without any apparent issues or errors. However, during evaluation, the agent's actions appear nonsensical and do not lead to successful task completion. This results in consistently negative evaluation metrics. I have also tried with a fresh environment, observing the same behavior.

**Environment:**

* **robomimic Version:** 0.4.0
* **robosuite Version:** 1.5.1
* **Python Version:** 3.13

amandlek, 2025-04-13:
Thanks for bringing this to our attention. Can you please give a specific example for us to try and reproduce the issue? e.g. A specific dataset link, and specific config (and details on how you obtained it). Thanks!

PartyPenguin, 2025-04-13:
Thank you for helping men.
I have installed a clean conda env as described in the documentation. 
Then I used this `python download_datasets.py --tasks sim --dataset_types ph --hdf5_types low_dim`  to download the datasets.
After that I used `python generate_paper_configs.py --output_dir /tmp/experiment_results` to get the config files.
Then I used `python robomimic/scripts/train.py --config robomimic/exps/paper/core/lift/ph/low_dim/bc.json --dataset datasets/lift/ph/low_dim_v15.hdf5`

As you can see in the script args, I am using plain bc with the lift task. Let me know if you need more Information. I also see this for my custom model that worked previously.

amandlek, 2025-04-13:
I tracked down the issue - I assumed that the low-dim observations would be the same between robosuite versions but it appears that this is not the case - observations should be re-extracted using the newer robosuite version. You can use a command like this: `python dataset_states_to_obs.py --dataset ../../datasets/lift/ph/low_dim_v15.hdf5 --output_name new.hdf5 --done_mode 2` and then update your `train.py` command to use the new dataset (`new.hdf5`). 

In the meantime, we will re-upload new postprocessed datasets to huggingface. Thanks a lot for your help in tracking this issue down!

amandlek, 2025-04-13:
The new datasets are now available to download on HuggingFace - please remove any existing datasets and re-download, and everything should work as expected now.

closed 2025-04-13 by amandlek

WhyxyxIsNotAnAvailableName, 2025-07-17:
> You need to download the new datasets. Robosuite has switched the way poses are returned in the observation. For instance for an object, it used to be absolute pose || relative pose to eef Now it's the other way around in the observation vector. Thats why the old dataset does not work with new robosuite version. So the easy fix is just to download the new dataset.

 Hello @PartyPenguin , could you please tell me where I can download the latest datasets? Using the codes in the documentation, such as: 
`python -m robocasa.scripts.download_datasets --ds_types human_im
`
the downloaded datasets are still the 1.4.1 version.

PartyPenguin, 2025-07-31:
> Hello [@PartyPenguin](https://github.com/PartyPenguin) , could you please tell me where I can download the latest datasets? Using the codes in the documentation, such as: `python -m [...]

Sorry for the late reply. In case you have not figured it out yet:

You need to use the recent release of robomimic. 
Follow this description: https://robomimic.github.io/docs/datasets/robosuite.html