ARISE-Initiative/robomimic #231: Option to disable Rollout Video saving during training

SurafelAnshebo, 2025-04-13:
I was wondering if there’s a way to disable saving the rollout video during training. I tried editing the JSON file, but it doesn’t seem to work. I believe turning it off could significantly help reduce the training time. Say if we are concerned about checking the success rate only

PartyPenguin, 2025-04-13:
I have not tried it but there is an option:  "render_video": true inside the json. 
Sounds, like it's the thing you're looking for.

amandlek, 2025-04-13:
Yup it should be `config.experiment.render_video` - does setting that to False not work for you?

closed 2025-04-22 by amandlek

SurafelAnshebo, 2025-04-26:
Hi @amandlek, sorry for the late update!
Turning off render_video alone wasn’t working for me, but I was able to stop Robomimic from writing rollout videos during training. I modified the config json file by setting "render_video": false, "render": false, and also updated "keep_all_videos": false just to be safe.

At first, I tried setting "n": 0 in the rollout section to skip evaluation episodes entirely (since it was set to 50 and taking a lot of time). But that caused an error because my understanding is the system expects at least one rollout to calculate metrics like success rate. So to fix it, I set "n": 1 instead, which now runs a single evaluation episode per checkpoint - this way, no video is written, but we still get the success rate and return values printed.

It seems to be working now, but I’m not 100% sure if this is the best long-term solution or if I did it right

amandlek, 2025-04-26:
Thanks for the additional details - if you want to disable rollouts entirely you could just disable them by setting [this](https://github.com/ARISE-Initiative/robomimic/blob/59ecd4048b91595b61cbcc2f3e4d9d666e4d4c04/robomimic/config/base_config.py#L118) to False.