ARISE-Initiative/robomimic #233: How can I figure out which cameras are available in an environment?

PartyPenguin, 2025-04-21:
I am currently trying to figure out which cameras are available in an environment. 
Is there a list? I could not find find any information on that. 

Right now I found the agentview and robot0_eye_in_hand camera as mentioned in the docs. There is also a shoulder camera mentioned. Are there any more cameras?

And a second question. I have this config:

```
...,
                "rgb": [
                    "agentview_image",
                    "robot0_eye_in_hand_image"
                ],...

```
The agentview is available during training. However during inference I don't get the agentview in my obs_dict. Is this camera not provided in the environment observations? I could not find any information on that as well.

I appreciate your help :)

amandlek, 2025-04-21:
The cameras available in a simulation environment depend on the simulation used - I'm assuming you're using robosuite - it might be better to raise this issue on their github page since they can provide more help on getting a list of cameras.

As for why the agentview is available during training but not during inference, it's possible that somehow the environment kwargs in the metadata are not setup properly - is the agentview camera listed in the env kwargs when running `get_dataset_info.py`? 

Another quirk is that in the robosuite wrapper class we use (`EnvRobosuite`) we rename all camera observation keys, so instead of `agentview` the observation is called `agentview_image`.

PartyPenguin, 2025-04-21:
Thanks for clarification.
When using `get_dataset_info.py` I do get the `agentview` listed.
And yes, I know that they get listed with `_image` appended. But when I look into the `obs_dict` it does not appear. 
I will ask in the robosuite to get some more insight :) 
Thanks again