ARISE-Initiative/robomimic #234: Correlation between action and the pos and quat of eef in the dataset

wudier483, 2025-04-22:
Hello, I am using the dataset provided by robomimic. But I don't understand relative control. I want to know how to get the pos and quat of eef at the next moment based on the current action and the pos and quat of the current eef. Looking forward to your answer.

PartyPenguin, 2025-04-22:
Yes, in theory you just need to add the predicted action to your current eef position. However the dataset uses some normalized actions, so you can't just add them together. 
I think control in general is handled by robosuite and not robomimic. So you might find some information on the particular controller/normalization in the other repository.

wudier483, 2025-04-22:
Thanks for your reply. I went to robosuite and found the config file corresponding to the controller:
{
"type": "OSC_POSE",
"input_max": 1,
"input_min": -1,
"output_max": [0.05, 0.05, 0.05, 0.5, 0.5, 0.5],
"output_min": [-0.05, -0.05, -0.05, -0.5, -0.5, -0.5],
"kp": 150,
"damping_ratio": 1,
"impedance_mode": "fixed",
"kp_limits": [0, 300],
"damping_ratio_limits": [0, 10],
"position_limits": null,
"orientation_limits": null,
"uncouple_pos_ori": true,
"input_type": "delta",
"input_ref_frame": "base",
"interpolation": null,
"ramp_ratio": 0.2
}

However, for Robomimic, do I need to change input_ref_frame to 'world'? 
Because I found this sentence in the document: "OSC_POSE: Gripper moves sequentially and linearly in x, y, z direction, then sequentially rotates in x-axis, y-axis, z-axis, relative to the global coordinate frame."

After combining it with the robosuite code, it means that action will be scaled before input. The first three dimensions (that is, the translation vector) need to be divided by 20, and the last three dimensions of action (corresponding to the rotation vector) need to be divided by 2.
The formula is: eef_pos[t+1]=eef_pos[t]+action[t]/20, but from the data, I can't get the correct eef_pos.

If I understand something wrong, please correct me. This is very important to me.
The following figure shows the action and eef_pos of a certain robomimic data.

[image]

PartyPenguin, 2025-04-22:
Achieving your goal can be quite challenging. I attempted something similar in a different simulator, and it didn't work as expected either.

The core issue is that the end-effector (EEF) position isn't simply the current position plus the action delta. When you send a target pose to the controller, it's important to understand that the controller does not guarantee reaching that target exactly. It's influenced by various factors like friction and other system losses.

Another complication arises from the way actions and observations are sampled in the dataset—typically at a fixed frequency (e.g., in Hz). Imagine your current position is (0, 0, 0), and your action is (1, 1, 1), suggesting a target of (1, 1, 1). While the controller will start moving toward that point, there's no guarantee it will arrive there within the same timestep that your observation is recorded.

For example, if the controller needs 1.3 timesteps to reach (1, 1, 1), but you're sampling at 1 timestep intervals, your observation may capture the robot at an intermediate position—not the final target. This discrepancy can cause mismatches between the intended motion and the actual recorded data.

closed 2025-04-22 by amandlek

2292685327, 2025-04-23:
Thank you, I have a few more questions:
1. Does this mean that the actions, eef_pos, and eef_quat contained in the data are based on the world coordinate system, not the robot base coordinate system?
2. The following is the content of a certain Robomimic dataset, with actions on the left and eef_pose on the right. Although the values of the first three dimensions of the action in the first frame are 0, does the eef_pose still change in the next moment?

[image]

3. If I use the absolute position as the action, will the action be the same as eef_pos and eef_quat?

The above questions are all due to my lack of understanding of data formats, and I did not find any relevant content in the document. Did I miss something?

amandlek, 2025-04-24:
(1) the `eef_pos` and `eef_quat` you're referring to are probably from the observations, not the actions, and in this case it is likely in the "world" frame

(2) yes it is possibly for the action to be 0 and for the end effector position to change - this is because the end effector robot controller is not perfect, and might not be able to perfectly achieve the target it is given

(3) no, the controller in general is imperfect, and will not achieve the exact pose that you give it