ARISE-Initiative/robomimic #236: Object observations mixed up in robomimic dataset?

PartyPenguin, 2025-05-05:
Apparently there have been changes to the order of the features in the dataset. The images are from demo_10 -> obs -> object out of the robomimic ph can dataset. Is there any information regarding the new structure? I used to rely on the information in the robomimic paper.

low_dim_v141.hdf5
[image]

low_dim_v15.hdf5
[image]

PartyPenguin, 2025-05-05:
Thanks for the clarification. Is there a documentation of the different features besides the information in the paper?

amandlek, 2025-05-05:
I don't think so, but it's worth raising an issue on the [robosuite](https://github.com/ARISE-Initiative/robosuite) github page.