ARISE-Initiative/robomimic #238: robomimic cannot open the file with human demonstration collected with robosuite

gokceay, 2025-05-19:
Hi, 

I collected human demonstration by using collect_human_demonstrations.py file from robosuite . Eventhough I can replay them with playback_demonstrations_from_hdf5.py with robosuite, robomimic cannot open the file with human demonstration collected with robosuite.  

It gives this error

"Traceback (most recent call last):
  File "/media/gokce-guven/Phoebe10TB/cobot/robosuite/robomimic_getstarted.py", line 186, in <module>
    model = get_example_model(dataset_path, device=device)
            ^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^
  File "/media/gokce-guven/Phoebe10TB/cobot/robosuite/robomimic_getstarted.py", line 68, in get_example_model
    shape_meta = FileUtils.get_shape_metadata_from_dataset(
                 ^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^
  File "/media/gokce-guven/Phoebe10TB/cobot/robosuite/robomimic/robomimic/utils/file_utils.py", line 156, in get_shape_metadata_from_dataset
    initial_shape = demo["obs/{}".format(k)].shape[1:]
                    ~~~~^^^^^^^^^^^^^^^^^^^^
  File "h5py/_objects.pyx", line 54, in h5py._objects.with_phil.wrapper
  File "h5py/_objects.pyx", line 55, in h5py._objects.with_phil.wrapper
  File "/home/gokce-guven/anaconda3/lib/python3.12/site-packages/h5py/_hl/group.py", line 357, in __getitem__
    oid = h5o.open(self.id, self._e(name), lapl=self._lapl)
          ^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^
  File "h5py/_objects.pyx", line 54, in h5py._objects.with_phil.wrapper
  File "h5py/_objects.pyx", line 55, in h5py._objects.with_phil.wrapper
  File "h5py/h5o.pyx", line 241, in h5py.h5o.open
KeyError: 'Unable to open object (component not found)'
"

What should I do?

gokceay, 2025-05-19:
maybe that is due to I cannot close properly the collect_human_demonstrations.py when the data is collected. How to safely exit the code when the data is collected?

amandlek, 2025-05-19:
I'm guessing that your dataset is missing observations - please take a look at the instructions in the link [here](https://robomimic.github.io/docs/datasets/robosuite.html) to solve your issue.

closed 2025-05-19 by amandlek