ARISE-Initiative/robomimic #239: Custom robot training was not successfull

gokceay, 2025-05-20:
I integrate my robot model Mycobot inside robosuite, and created human demonstration data from robosuite and then converted it to robomimic dataset . I used the code below for training: 

https://colab.research.google.com/github/AssistiveRoboticsUNH/bc_tutorial/blob/main/robomimic_tasks/train_lift_minimal.ipynb

Eventhough I trained successfully with 280 demonstrations and the loss value was decreasing none of my rollouts was successfull. 

What could be the reason for this?

Any help could be appreaciated.
Thanks in advance

amandlek, 2025-05-20:
It's hard to pinpoint exactly what your issue might be. Some follow-up questions:
1. Did you visualize your dataset, and also get statistics on how long the trajectories are? You should take care to tune your rollout horizon accordingly.
2. What were the success rates like during the evaluation rollouts that occur during training? If you're only looking at the training loss, there's no guarantee that rollouts will be successful from a low training loss

gokceay, 2025-05-21:
Thanks a lot for your fast response 

1. I visualized my dataset and mean trajectory length is 203 and as horizon I used 400. 
2. my success rate is 1. "{'Return': 1.0, 'Horizon': 400, 'Success_Rate': 1.0}"

But still the robot is not grasping, it is hard for me to understand what I am missing

gokceay, 2025-05-27:
First I have tried with colab but that did not worked for my case. Then I switched to the normal training (train.py) and evaluation (run_trained_agent.py). After training 4000 epoch with BC RNN my success rate was 0.77 (low_dim.hdf5). I had 200 human demonstrations collected with keyboard. Now I increased the number of human demonstrations to the 500 and created image.hdf5 for a new training. For the config, I started from bc.json, enabled both RNN and GMM, increased the RNN hidden dimension from 400 to 1,000, and set seq_length to 10. I’m now waiting on the results. I’ll dive deeper into the paper for further hyperparameter tuning. Thanks again for your help