ARISE-Initiative/robomimic #270: Conversion from absolute actions to delta actions

vincenzopomponi, 2025-09-05:
In the latest release of Robomimic v0.5.0, the script [robomimic/scripts/conversion/convert_robosuite](https://github.com/ARISE-Initiative/robomimic/blob/main/robomimic/scripts/conversion/convert_robosuite?utm_source=chatgpt.com)
 introduced support for converting datasets from delta actions → absolute actions.

It would be very useful to also support the inverse operation: converting absolute actions → delta actions.

Motivation

Having this functionality would make it easier to compare methods that use different action representations, and would significantly simplify code development for benchmarking across approaches.

Proposal

Add an option to convert_robosuite (or a new script) to transform datasets with absolute actions into delta-action datasets.

Ideally, the interface and usage could mirror the current implementation for the delta → absolute conversion.

Thanks for considering this feature!

akushonkamen, 2026-10-17:
Taking this one (per @amandlek's green light above). Plan: add `robomimic/scripts/conversion/robosuite_add_delta_actions.py` mirroring `robosuite_add_absolute_actions.py:38` (RobomimicAbsoluteActionConverter) — a `RobomimicDeltaActionConverter` that resets to each stored state, reads the current eef pose from the OSC controller, and inverts the controller's delta->goal map (`robosuite/controllers/osc.py`) analytically to recover per-step delta actions from absolute actions, writing them under a new `actions_delta` key (trainable via `config.train.action_keys`, see `robomimic/config/base_config.py:208`). Will include a round-trip regression test (delta -> abs -> delta identity on the small HF test dataset).

(Note: I'm an AI-assisted contributor working on this; all reproduction and testing will be run locally.)

akushonkamen, 2026-10-18:
Opened PR #326 implementing the absolute → delta conversion (mirrors `robosuite_add_absolute_actions.py`, writes an `actions_delta` key per demo). Round-trip test `tests/test_conversion.py` (delta → abs → delta) passes with max error 4.1e-08 on `tests/assets/test_v15.hdf5`.